xoseperez/hlw8012 #13: HLW8012 CP1 frequency capping

TrCaner, 2019-04-16:
I'm currently designing a circuit that contains the HLW8012.
But I'm having a problem with the CF1 pin capping at 1.7kHz output, while in current mode.
The circuit I use is the one provided in the datasheet of the HLW8012, with the MCU connected directly to the HLW8012 pin.
I have tried replacing components in case it was a bad component, but no luck so far.
Since non of this worked I tried to start with a low current and increasing it to see where it caps. 

My results were as follows:
0.323A -> 348Hz
0.523A -> 561Hz
0.716A -> 753Hz
1.256A -> 1332Hz
1.563A -> 1602Hz
1.673A -> 1724Hz
Everything above the 1.6A gets a frequency of around 1700~Hz (1.7kHz).
(For the load, I used multiple lamps that consume between 50~120W)

Any idea's what could cause this problem?

uzi18, 2019-08-05:
@TrCaner did you used multimeter/frequency meter?
please paste your test code

TrCaner, 2019-08-06:
I used 3 methodes to measure the frequency. The first was a multimeter, second a oscilloscope and last a simple frequency measurement code that makes uses of PCNT, timers, interrupts etc. All 3 gave the same reading of the frequencies. 

To be sure I ordered a new batch of chips from a different supplier and this solved my problem. I could now accurately measure the current (tested to 8A). This meant that I had ordered a fake set of chips the first time that couldn't measure more than what i stated before. 

Since the problem is solved I'll be closing this issue.

closed 2019-08-06 by TrCaner