xoseperez/hlw8012 #16: Multiple HLW8012

abhi-ratnman, 2019-07-11:
Hey Xose, I am working on a Home Automation product which will be able to send Energy Measurement using this HLW8012 module (Link: https://www.electrodragon.com/product/energy-meter-hlw8012-breakout-board/). Thanks to you we were able to work with 1 Module. The problem comes when we try to connect multiple modules with same MCU giving them power from an isolated source(Hi-link HLK-PM01) the problems is when we switch on one HLW8012 the other HLW8012 module which is connected to power source (Hi-link HLK-PM01) but the switch is off still gets the AC power which in-turn switch on the device.

Please Help us with guiding for developing a module which is able to Get power reading from 4 different Appliances using the same isolated power source.

Thanks,
Abhi Ratnman
homeautoma987@gmail.com

Oushi157, 2019-08-03:
Hi Abhi. Sorry I don't know the solution to your issue but I am stuck at what you have already achieved. I too am using the breakout board, but am only getting zero readings. I tried both the basic.ino as well as the interrupt.ino from this library. Is this me only or you also faced issues with the board? 
I have attached images of my setup and the readings I get.
[image]
[image]

abhi-ratnman, 2019-08-14:
Contact us on homeautoma987@gmail.com