xoseperez/hlw8012 #17: Zero Readings

Oushi157, 2019-08-03:
I am using the HLW8012 Energy meter module, and have used the HLW8012 Basic.ino from this library. However the readings are always zero for all measurements when the load is on. Could this be an issue with reading pulse from the module? Or if anyone has tried this code on the module before? 
https://www.electrodragon.com/product/energy-meter-hlw8012-breakout-board/

Urs-Eppenberger, 2020-09-06:
For tests I suggest to use an old light bulb which is a purely resistive load. Or a heater.
The zeroes you get might also be just the fact, that your led bulb only uses current of several mA.

joshuaboniface, 2021-02-19:
Are you trying to use the Interrupt example? I'm finding that this just does not work, nor does calibration. If I use the non-interrupt example it works.

joshuaboniface, 2021-02-19:
I found, at least in my case, is that the Interrupt example has not been updated (I presume) for how the newer Arduinos work.

As per [the Arduino documentation on the `attachInterrupt` function](https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/), when doing an `attachInterrupt`, the digital pin ID itself should not be set, as the example does:

```
  attachInterrupt(CF1_PIN, hlw8012_cf1_interrupt, CHANGE);
  attachInterrupt(CF_PIN, hlw8012_cf_interrupt, CHANGE);
```

Instead, the `digitalPinToInterrupt` function should be used to wrap the pin ID, in order to send the proper interrupt:

```
  attachInterrupt(digitalPinToInterrupt(CF1_PIN), hlw8012_cf1_interrupt, CHANGE);
  attachInterrupt(digitalPinToInterrupt(CF_PIN), hlw8012_cf_interrupt, CHANGE);
```

This appears to work for me.

Also note that, at least on the Arduino Uno, only digital pins 2 and 3 support interrupts; the full list can be found in the above documentation.

xoseperez, 2021-02-19:
Thanks @joshuaboniface , I updated the examples with your code!