xoseperez/hlw8012 #19: Voltage and current power formula

1404360768, 2020-04-11:
Hello, I don't understand where "1000000.0" in the calculation formula comes from, the value I calculated with the formula and the deviation I measured with the power meter。

    _current_multiplier = (1000000.0 * 512 * V_REF / _current_resistor / 24.0 / F_OSC);
    _voltage_multiplier = (2000000.0 * 512 * V_REF * _voltage_resistor / 2.0 / F_OSC);
    _power_multiplier = (1000000.0 * 128 * V_REF * V_REF * _voltage_resistor / _current_resistor / 48.0 / F_OSC);

xoseperez, 2021-02-19:
This is due to the library getting the period from the sensor in microseconds, so frequency is not in Hz (as per datasheet) but in 1/us. Hence the 10^6 in the multipliers (to avoid doing the calculation afterwards).

1404360768, 2021-02-20:
Thank you, I see!