xoseperez/hlw8012 #24: HLW8012 board calibration

pw44, 2021-02-16:
seams this one is also becoming popular. 

but i'm not find out to calibarte it. any help?

Thx in advance.

[image]

xoseperez, 2021-02-19:
You have a calibration example in the "HLW8012_Interrupts" example. It returns the multipliers you can use on library setup with the setCurrentMultiplier, setVoltageMultiplier and setPowerMultiplier.

pw44, 2021-02-19:
i used the HLW8012_Basic example, with no luck. 

By calibrate section, i setted:

400W (my toaster)
127V (my AC mains, according my MM
and 400/127

`   // Calibrate using a 60W bulb (pure resistive) on a 230V line
    hlw8012.expectedActivePower(60.0);
    hlw8012.expectedVoltage(230.0);
    hlw8012.expectedCurrent(60.0 / 230.0);`

But output does not shows 127, 400, etc....